STM32-base/STM32-base.github.io #9: boards/STM32F746IGT6-WaveShare-Core7XXI.md: SDRAM is 8 Mb, not 64 Mb

abelykh0, 2019-10-14:
For this board SDRAM is 8 Mb, not 64 Mb (see http://www.issi.com/WW/pdf/42-45S16400J.pdf). ^4 MBit = 8 Mb

ThomasGravekamp, 2019-10-14:
Thanks for submitting this issue.

I always thought that MB meant Megabyte and that Mb meant Megabit. I guess I got that from [data transfer rates](https://en.wikipedia.org/wiki/Data-rate_units), in which Mb seems to be used more commonly.

I will fix this.

abelykh0, 2019-10-14:
Maybe you are right, however if you google "64 mbit in mb" it will show 8 "Megabyte"

ThomasGravekamp, 2019-10-14:
If you want you can include the fix for this in your current PR. You may change the text to "64Mbit".

abelykh0, 2019-10-14:
Let's keep it separate, thanks.

abelykh0, 2019-10-14:
I would actually prefer to see 8 megabytes, but it is your site...

ThomasGravekamp, 2019-10-15:
Hm, yeah, I can agree on that, to keep it consistent with the memory size units used for the microcontroller itself.

ThomasGravekamp, 2019-10-20:
Done, thanks for your input.

commit c91c08ab (2019-10-20): Use MB to quantify the external memory's size on the Waveshare Core7XXI board page. Fixes #9.
closed 2019-10-20 by ThomasGravekamp