STM32-base/STM32-base.github.io #12: Add STM32H7 board

abelykh0, 2019-11-25:
Add STM32H7 board, such as https://www.ebay.ca/itm/STM32H750VBT6-Development-Board-ARM-STM32H7-Programmable-Controller-Module/254400595200?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m1438.l2649

ThomasGravekamp, 2019-11-27:
I'm not sure what your intentions with this issues are. Are you asking me to add this exact board or just any board featuring an STM32H7 chip? Or are you just pointing out to me that these boards exist? (Which I didn't by the way).

I've still got 12 unique boards sitting on my desk for which I have to make a page, so it might take some time to get around to a board like this.

abelykh0, 2019-11-28:
It is not really an "issue" (it is a "wish" or a "feature request"). Yes, it would be nice to add any boards featuring STM32H7 chip, if possible. Obviously, you can take as much time as you want (or non at all) for this.

ThomasGravekamp, 2019-12-30:
Status update: The first boards have arrived!

commit 3015a22b (2020-06-13): Add STM32H743VIT6 and STM32H750VBT6 STM32H7XX_M board pages. Fixes #12.
closed 2020-06-13 by ThomasGravekamp