STM32-base/STM32-base.github.io #15: Add a "Connect a display" page

ostanki, 2019-12-03:
A "Connect a display" page might be nice. Contents/keywords:
Some basics about displays: Types (OLED, TFT, text, graphic...), display controller(s)
Types of displays as seen from an STM32: I2C ,SPI, Parallell, RGB ... Necessary signals (examples?)
"Cheating" by using FSMC to connect a parallell display, signals in a (FSMC) TFT connector (example?)
Some STM32s even have an onboard LCD controller: limitations, uses...

You can assign it to me, but it will take time, and I will probably need help/guidance...

ThomasGravekamp, 2019-12-03:
Looks good!

I have assigned this issue to you. You can take your time on this one. The amount of work also depends a little bit on where you want to take this page.

For pages like these I want to introduce "Concept pages". The different between guides and concept pages is that guides will help you achieve one specific goal and that concept pages contain a lot more specific/broader information but do not "guide" you through a specific process.

The page you are currently proposing should be a concept page. The only thing that currently suggests that it should be a guide is the title. You could start out by describing how the FSMC can be used to control a display and work from there. That in itself would be enough to publish the page. Then you can expand the page later. But I'll leave this to you.

Just ask for help when you need it.

ThomasGravekamp, 2020-11-06:
Closing this issue due to inactivity.

closed 2020-11-06 by ThomasGravekamp