STM32-base/STM32-base.github.io #35: Add STM32_F030 board

allenchak, 2020-12-17:
Board Name: STM32_F030
Inputs, outputs, connectors & headers:

1. STM32F030F4P6
2. MicroUSB connector
3. SWD connector
4. P1
5. P2
6. LED: 2 (D1: Power LED; D2: User LED, PA4)

- D1: Power LED
- D2: User LED, PA4

7. Button: 2

- K1: User Button, PA0
- RST: NRST

8. Schematic included

[image]
[image]
[image]
[image]

ThomasGravekamp, 2020-12-17:
Thanks for bringing this to my attention.

Did you take that first picture yourself? If so, would you be able to also take one of the back of the board and one that has a perspective view.

The pictures on stm32-base.org are licenced under the Creative Commons Attribution-NonCommercial 4.0 International (CC BY 4.0), so I can not randomly use someone's pictures. They'll have to agree to share them.

allenchak, 2020-12-17:
Yes, the image1 was taken by me. Sure, I could take 2 more images to you, but I have a sticker on the backside, I try to remove it. Or else you do not mind?

allenchak, 2020-12-17:
Unfortunately the stickers quite large, but lucky it does not cover any silkscreen.
Anyway, I will upload the image tomorrow. If you need the product link, just let me know.

allenchak, 2020-12-18:
[image]
[image]

ThomasGravekamp, 2020-12-20:
Thanks! I'll create a page for this board when I find the time for it somewhere the coming two weeks.

martinstingl, 2020-12-24:
Please also add the schematic from the initial post to the board page, most STM32F030F4P6 seems to use it.

ThomasGravekamp, 2020-12-29:
@martinstingl Can you please verify that the connections listed for the buttons and LEDs are correct? I looked at the schematic, but it looks like it is meant for another board.

closed 2020-12-29 by ThomasGravekamp

rfivet, 2021-01-02:
This is the droid you are looking for.
[image]

martinstingl, 2021-01-12:
@ThomasGravekamp
You are right, I measured the connection on this board and there is a difference:
https://stm32-base.org/assets/img/boards/STM32F030F4P6_STM32F030_DEMO_BOARD_V1.1-2.jpg

- PA4 is connected directly to the LED, its anode is connected to 3V3 via 1k
- BOOT0 is pulled to 3V3 via 1k, and can be connected to GND or 3V3 with the 3 position jumper

- The reset button is connected unchanged

ThomasGravekamp, 2021-01-16:
Thanks for verifying @martinstingl, looks like I've made no mistakes.

commit 5ed00c06 (2020-12-29): Add STM32_F030 board page. Fixes #35.