STM32-base/STM32-base.github.io #37: WeAct Black Pill V3.0 and direct power warning

jej, 2021-01-03:
Hi,
The V2 page of this board says: **_Warning: The +5V pins on this board are directly connected to the +5V pin of the USB connector. There is no protection in place. Do not power this board through USB and an external power supply at the same time._** ([here](https://stm32-base.org/boards/STM32F411CEU6-WeAct-Black-Pill-V2.0.html))
V3 page says the same, but the schematic includes a protection diode ([here](https://stm32-base.org/assets/pdf/boards/original-schematic-STM32F401CEU6_WeAct_Black_Pill_V3.0.pdf))
is it an update error ported from the V2 page?

[image]

ThomasGravekamp, 2021-01-03:
Yup, looks like something that has been copied over to the V3 page.

ThomasGravekamp, 2021-01-03:
Double checked the V2 page, seems like an improvement to the V3 board. Fixing it now.

commit ccc12152 (2021-01-03): Remove statement from WeAct Black Pill V3.0 board page that it does not have protection on the +5V rail. Fixes #37.
closed 2021-01-03 by ThomasGravekamp