STM32-base/STM32-base.github.io #41: SongHe stm32f401CCU6 board

vsessink, 2021-03-26:
Hoi Thomas, hi others, found a new board. Please see https://www.amazon.com/SongHe-STM32F401-Development-STM32F401CCU6-Learning/dp/B07XBWGF9M and https://www.electrodragon.com/product/stm32f401-mini-dev-board-stm32f401ccu6/ with more pictures.

The board is similar to Black Pills, but has a few differences: no user button; BOOT0 and BOOT1 are dip switches.

Even more information:
https://www.electrodragon.com/w/STM32F4
https://drive.google.com/file/d/1I3E_-ZKEtSfsHiNDR-I87v4AHSIqD978/view (schematics)
https://cntronic.com/stm32f401-development-board-stm32f401ccu6-stm32f4-learning-board-30332 (still more pictures)

The following pictures can be used under public domain, I made them and I'm donating them. I can make some more pictures with a neutral grey background if that is what you need.

@ThomasGravekamp ik heb een pull request geprobeerd, maar welke power regulator enz - ik weet het echt niet. Laat weten wat je nodig hebt. Geweldig project, trouwens!

[image]
[image]

ThomasGravekamp, 2021-04-27:
Hi, thank you for wanting to contribute!

I prefer having pictures with a grey background to keep them consistent, if that's not too much trouble. I take three pictures of each board: a perspective view (shown on the boards overview page), a top view, and a bottom view.

I'll have free some time to add this board page. By the way, the regulator used on this board is the XC6204: https://stm32-base.org/assets/pdf/regulators/XC6204-XC6205.pdf.

vsessink, 2021-04-29:
[image]
[image]
[image]

ThomasGravekamp, 2021-05-05:
I've added the board page. Thanks for helping out with the images!

commit e89c350f (2021-05-05): Add STM32F401CCU6 STM32 Mini F401 board page. Fixes #41.
closed 2021-05-05 by ThomasGravekamp