STM32-base/STM32-base.github.io #45: STM32F103C8T6 Blue Pill Schematic Error

emrelio, 2022-04-29:
Hi! 

First off all your page is very informative and easy to use. Thank you for that.

In the "STM32F103C8T6 Blue Pill" board schematic, there is an error with the main 8mHz crystal. 20pf caps must have connected to the ground. With the current design, the external crystal won't work.

Thank You!

ThomasGravekamp, 2022-05-23:
Hi, thank you for reporting this issue.

I do not have the ability to make changes to that schematic. Do you know whether there is a newer version available on the internet somewhere? If not, I'll have to add a note on the Blue Pill board page that the schematic is not entirely correct.

emrelio, 2022-05-23:
I don't have any updated schematic. This schematic is everywhere in the internet with this error. But that would be nice to warn people about that.

closed 2022-05-23 by emrelio